ut-robotics/picr22-team-owo #20: MCU pin assignement

ReikoR, 2022-10-26:
[image]

I could not find working pin usage for the MCU.

Can you explain how the timer channels are used?

Most important part is that PWM outputs can't be properly used on timers that have encoder mode enabled, as in encoder mode the counter value is changed based on encoder inputs.

dakoko, 2022-10-29:
I was unaware of the encoder mode and PWM output conflict. I changed the timers and pins around so that they'd be seperated from now on (new and old timer/pinout/channel setup is brought out in the pinout documentation spreadsheet file).

Timers 1, 3 and 4 are set to encoder mode. Timer 2 handles the thrower motor PWM, timer 8 the 3 motor PWMs and time 15 the servo PWMs.

For this fix I had to sacrifice the PWM input meant for the opto sensor and was replaced with an ADC. I will most likely handle the smoothing and other possible complications offboard.

ReikoR, 2022-10-30:
For optical sensor it's probably best to use EXTI (external interrupt) as the sensor output is digital.
[image]

Otherwise new pin assignment might work.

closed 2022-10-30 by ReikoR