ut-robotics/picr22-team-owo #32: PA1 pin used for ball sensor is not 5V tolerant

ReikoR, 2022-11-06:
[image]

From [datasheet](https://www.st.com/resource/en/datasheet/stm32g441kb.pdf):
[image]
[image]

dakoko, 2022-11-06:
The output signal of the ball sensor is going to be brought down to 3.3v with the offboard electronics needed for the sensor. Most likely through a voltage divider.

ReikoR, 2022-11-06:
It would be easier to use one of the 5V tolerant pins. For example PF0 or PF1:
[image]

dakoko, 2022-11-06:
Switched the signal led pin (PF1) with the ball sensor pin (PA1)