ut-robotics/picr22-team-owo #37: Place GND vias close to components

ReikoR, 2022-11-09:
For example:
[image]

dakoko, 2022-11-09:
Added lots of vias.

ReikoR, 2022-11-09:
[image]

dakoko, 2022-11-09:
Suggested locations are directly above 2 power rails, is it really worth putting a hole through VBAT?
[image]

ReikoR, 2022-11-10:
5V track can be moved. Having it next to VBAT is not good anyway.

Something like this:
[image]

dakoko, 2022-11-10:
Moved the trace

closed 2022-11-10 by ReikoR