ut-robotics/picr22-team-owo #41: Improve switching regulator layout

ReikoR, 2022-11-09:
Try to follow guidelines in the [TPS62933 datasheet](https://www.ti.com/lit/ds/symlink/tps62933.pdf) under 12.1 Layout Guidelines.

1. C6 capacitor should be at the output:
[image]

2. VIN and GND connections should be wider and with more vias.
\
From datasheet:
The VIN and GND traces must be as wide as possible and provide sufficient vias on them to minimize trace impedance. The wide areas are also of advantage from the view point of heat dissipation.
\
[image]

3. SW connection should be wider.
\
From datasheet:
Keep the SW trace as physically short and wide as practical to minimize radiated emissions.
\
[image]

4. Route feedback trace further away from the switching trace. In the current layout it's hard to route it through bottom layer, but it can be routed a bit farther from the inductor. It should also connect to output capacitors.
\
From datasheet:
Place the voltage feedback loop away from the high-voltage switching trace. The voltage feedback loop preferably has ground shield.
\
[image]

dakoko, 2022-11-10:
Alleviated the issues, don't know if it's enough though.

ReikoR, 2022-11-10:
It's possible to make SW copper a bit wider by moving the regulator downwards.

Before:
[image]

After:
[image]

I recommend using Polygon Pour instead of Region. Polygons automatically pour based on the rules.

Polygons:
[image]

Regulator input GND polygon properties:
[image]

I created a Polygon Connect Style rule to use direct connection for these polygons:
[image]
`InNet('NetC13_1') or InNet('VBAT') or ((ObjectKind = 'Poly') And (Name = 'GND_SWREG_IN')) `

Pour order also needs to be changed in Polygon Manager (Tools -> Polygon Pours -> Polygon Manager):
[image]
Smaller polygons need to poured before larger GND polygon.

dakoko, 2022-11-10:
Replaced the regions with pours

ReikoR, 2022-11-10:
There are very thin gaps next to the GND polygon:
[image]

dakoko, 2022-11-11:
Added thermal relief and fixed the thin gap issue

ReikoR, 2022-11-11:
It would be better to have GND directly connected.

dakoko, 2022-11-11:
Connected it directly

closed 2022-11-11 by ReikoR